klaasnicolaas/home-assistant-glow #528: release 4.1.0 - Nodemcu systems cannot connect

peter656761, 2024-07-06:
I am a relatively new user of Homeassistant and Nodemcu boards but have  found the latest release 4.1.0 seems to have stopped working 

For example I can flash via cable a new NodeMCU esp8266.  It asks for wifi credentials that are input and it gets an ip address.
Going to add the unit to ESP home I notice that the included link is for the D1 mini not the esp.  I can leave it as a D1 or change it to the esp and either way then adopt the board and install.  This is the only time the logs are visible and I see that it connects to the homeassitant and then disconnects continually.

Upgrading produces the same result - a system seemingly installed but cannot be connected to

 
[image]

klaasnicolaas, 2024-07-06:
Which board exactly do you have?

./Klaas

klaasnicolaas, 2024-07-06:
After testing, I have also noticed that the D1 Mini firmware is being used, so I'm going to look for a solution for that. I wouldn't know of an immediate solution other than removing the device everywhere and trying to add it again.

./Klaas

klaasnicolaas, 2024-07-14:
Have you been able to find a solution yourself in the meantime?

./Klaas

peter656761, 2024-07-15:
Hi Klaas

Thank you for the email.  I had to manually initiate the Nodemcu install via ESPhome then adopt it into Homeassistant.

I then downloaded your  yaml and made a few modifications to work perfectly thank you

Regards
Peter#

closed 2024-07-15 by klaasnicolaas