klaasnicolaas/home-assistant-glow #536: Power consumption "spam"?

getninjaN, 2024-07-13:
Firstly, not sure what title to set. Sorry 'bout that.

My Glow has started acting up.
Updated to the latest version through the website after a Home Assistant update which renamed something, or something, and now the consumption has gone haywire.
Any ideas?
All the components are the recommended once in the README here on GitHub.

I tried changing the `internal_filter` setting but not sure if that made a difference.

https://github.com/user-attachments/assets/4e18779f-51f4-46cf-9761-e67c127f4d91

klaasnicolaas, 2024-07-13:
Can you also show your yaml config? Or the adjustments you made?

getninjaN, 2024-07-13:
> Can you also show your yaml config? Or the adjustments you made?

Ofc. This is my `home-assistant-glow-e7d3b4.yaml` file. Is this the one you need?
The thing I added was `internal_filter: 100ms` after it went crazy.

```yaml
substitutions:
  name: home-assistant-glow-e7d3b4
packages:
  klaasnicolaas.home-assistant-glow: github://klaasnicolaas/home-assistant-glow/home-assistant-glow/esp32.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
api:
  encryption:
    key: <key>

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

sensor:
  - id: !extend sensor_energy_pulse_meter
    internal_filter: 100ms
```

klaasnicolaas, 2024-07-13:
Do you know which version you were running before? Have you checked whether everything is connected to the correct GPIO pins?

./Klaas

getninjaN, 2024-07-13:
> Do you know which version you were running before the update? Have you checked whether everything is connected to the correct GPIO pins?
> 
> ./Klaas

Unfortunately I don't. My guess is 4.0.0 but it could be 3.1.0. Initially it was some time ago I set it up.
After updating to HA 2024.07.1 I noticed that the energy tab was showing 0 watts on all the days from that update.
So I updated Glow when I noticed thinking something changed somewhere and that's when I noticed it got all weird.

I haven't tried going back in versions yet, haven't had the time really, but I could try to do that if it helps.

I have checked the GPIO pins multiple times since my googling told me something about different pins and from what I can tell I have connected it all according to the current wiring diagram.

klaasnicolaas, 2024-07-13:
> After updating to HA 2024.07.1 I noticed that the energy tab was showing 0 watts on all the days from that update.

That could be a bug in Home Assistant with the recorder 😉 

Because your logs fill up so quickly, I have the feeling that this is due to a cable being connected to the wrong GPIO pin. This change is in release v4.0.0, hence my question about what you were running before the update.

./Klaas

getninjaN, 2024-07-13:
> That could be a bug in Home Assistant with the recorder 😉

Aaw maan. 🙂 

I'll try to check the pins again tomorrow, take some photos and come back to you asap if it works or not.

getninjaN, 2024-07-14:
Ok, I've checked and tried different sides and checked the wiring diagram. I'm still having issues.

This is my wiring:

## Status LED
[image]

## Diode
[image]

# The following did not work at all, it didn't even boot:
[image]
[image]

# The following booted, but is giving me the trouble that the issue is about
[image]
[image]

klaasnicolaas, 2024-07-14:
You are using D27 for GND and EN for 5v/3.3v?

./Klaas

klaasnicolaas, 2024-07-14:
I think you still have the wrong GPIO pin

[image]

getninjaN, 2024-07-14:
> I think you still have the wrong GPIO pin, you are using GPIO12.
> 

Holy shit, you're right ... 😮 
I never actually checked the f*cking pin numbers on the board, I just looked at the wiring diagram and counted ... 😫 

I feel like such an idiot. What a waste of time. 🥲 
Thanks, it seems to be working now!

getninjaN, 2024-07-14:
Oh, I'm closing this.

closed 2024-07-14 by getninjaN