klaasnicolaas/home-assistant-glow #538: Cannot connect to hidden SSID

deanfourie1, 2024-07-15:
I am trying to install GLOW on a d1_mini and connect it to my IoT WiFi.

This SSID is a hidden SSID and the entire network has no internet connection, Is installation of GLOW possible with this scenario.

Usually, with ESPhome devices, being a hidden SSID I have to manually specify what BSSID to connect to. But in this case, I cannot do this? Is there a workaround for this?

Thanks

klaasnicolaas, 2024-07-15:
I think the best solution is to create a node in ESPHome Dashboard and put in the YAML config from the repository, adjust it to what is needed to connect to a hidden WiFi network and then compile and install the software on your ESP.

./Klaas

deanfourie1, 2024-07-15:
Yes thanks I thought that too, however I see external resources are pulled inside the YAML.

Will those resources fail to download as I have no internet on this particular SSID.

klaasnicolaas, 2024-07-15:
If you have cut off your entire IoT network from the internet, you are making things quite difficult for yourself. If I push a new release, you won't get any updates, so you have to keep track of everything manually. Can't you allow certain resources? For example, GitHub and the manifest on the glow website?

Otherwise you also have to put all the yaml from the external resources in your config to make it complete.

./Klaas

deanfourie1, 2024-07-15:
I have combined all of the YAML from 

- basis.yaml
- pulse_meter.yaml
- status_led.yaml
- updates.yaml

Into one combined YAML. This is working, is this all the is required as external resources?

Thanks

klaasnicolaas, 2024-07-31:
> is this all the is required as external resources

Yes those are all external files.

./Klaas

deanfourie1, 2024-07-31:
Yes, thanks.

I just manually added the YAML as per the rest of my devices and all goo.

Thank you!

closed 2024-07-31 by deanfourie1