klaasnicolaas/home-assistant-glow #540: After first reboot D1 Mini : [http_request.arduino:132]	 Stream pointer vanished!

dirkclae, 2024-07-15:
I can install the firmware via USB. I can visit the webpage and everything shows correctly. Resettig the device or power off/on and it starts to throw Stream pointer vansihed. Tested with 2 different cards. Same behaviour.

klaasnicolaas, 2024-07-15:
This is a bug in ESPHome: https://github.com/esphome/issues/issues/5949

./Klaas

crcerdas, 2024-07-22:
There was a new Release of ESPhome today for HA. Anyone know if this has beeen resolved?

klaasnicolaas, 2024-07-22:
The previously mentioned issue is still open, so no 😉 

./Klaas

duchu, 2024-07-23:
Is there any workaround for this?

klaasnicolaas, 2024-07-31:
@duchu Using a different board type (ESP32 recommended)

./Klaas

epper202, 2024-10-13:
Hi!
I only seem to have this error after successfully installing the firmware whereafter I need to switch the power supply for the esp. After that it goes wrong and gives me pointer stream vanished 137
I have 1 glow on 4.0.1 which works perfectly, disconnecting and connecting psu. 
Yesterday I made 2 new ones and same issue. 
Is this still an Esp issue?

klaasnicolaas, 2024-10-13:
The last time I tested it worked fine and I could not reproduce the error anymore. I'm mainly waiting patiently until it upstream (ESPHome) is resolved, unfortunately I can't do more than that.

./Klaas

epper202, 2024-10-13:
Thanks for your reply, can I OTA the same firmware via the webbased IP?
Which file do I need to select, first time doing that actually but I was on a remote location and fighted with them the entire morning and gave up haha. much appreciated!

epper202, 2024-10-13:
> Thanks for your reply, can I OTA the same firmware via the webbased IP? Which file do I need to select, first time doing that actually but I was on a remote location and fighted with them the entire morning and gave up haha. much appreciated!

Found a warkaround.
step 1 flash firmware via your website
step 2 click next but in my case no connect to wifi appears, it's like it's being hold in a boot mode?
step 3 open console + logs instead and hit reset device and immediately it shows a lot of info
step 4 disconnect esp from laptop
step 5 place esp wherever you want
step 6 connect with phone/laptop via wifi point and add to the wifi point of choice.
step 7 done

epper202, 2024-10-15:
> > Thanks for your reply, can I OTA the same firmware via the webbased IP? Which file do I need to select, first time doing that actually but I was on a remote location and fighted with them the entire morning and gave up haha. much appreciated!
> 
> Found a warkaround. step 1 flash firmware via your website step 2 click next but in my case no connect to wifi appears, it's like it's being held in a boot mode? step 3 open console + logs instead and hit reset device and immediately it shows a lot of info step 4 disconnect esp from laptop step 5 place esp wherever you want step 6 connect with phone/laptop via wifi point and add to the wifi point of choice. step 7 done

So even this was a temporary fix it seems.. 
I've spend 2 days of trial and error, it works, disconnect the esp8266 d1 mini and reconnect it. bam, stream pointer vanished..
This is making me crazy haha

Phillip-VSA, 2024-11-07:
Did you ever get a way to fix it? I'm having the same problem and can't seem to figure out a way to get it to work.

epper202, 2024-11-07:
Yes. 
I got rid of the esp8266's. Bought a 5 pack esp32's changed the code to my needs for 3 meters and all works perfectly. 
Bottom line in my case it was the D1 mini which was fooling me.

Rayleigh3105, 2024-11-17:
> Found a warkaround. step 1 flash firmware via your website step 2 click next but in my case no connect to wifi appears, it's like it's being held in a boot mode? step 3 open console + logs instead [...]

Thanks this worked for me :)

karthiksogal, 2025-03-27:
is the issue still going on? I just started to face it in esp home 2025.3.2 with glow version 4.2.3, board esp 8266 nodemcu
[E]
[http_request.arduino:137]
Stream pointer vanished!
kindly help!