klaasnicolaas/home-assistant-glow #557: ESP32 board is crashing after first reboot

gezuka77, 2024-08-03:
Hi,

While I have my D1 mini adventures, I am also trying to use my ESP board(s). Flashing via the website went well ok. I can access the built in interface. I have added to Home Assistant via the discovery. Entities show up.

The issue:
However when I use the reboot button on the interface the ESP never comes back up. I don't have log messages either.
This also happens when I adopt in the ESP Home in Home assistant. That time it pushes an firmware update to the ESP board and it reboots. After that it also never comes back up.

I have 5 different ESPs. All behave the same.

gezuka77, 2024-08-03:
I now managed to get some serial messages to the ESP board:

```
time 0 mSec[0m
[0;36m[D][esp-idf:000][1;31m[wifi][0;36m: 
[0m
[0;36m[D][esp-idf:000][1;31m[wifi][0;36m: E (14822) wifi:[0m
[0;36m[D][esp-idf:000][1;31m[wifi][0;36m: Association refused temporarily, comeback time 0 mSec[0m
[0;36m[D][esp-idf:000][1;31m[wifi][0;36m: 
[0m
[0;36m[D][esp-idf:000][1;31m[wifi][0;36m: E (14842) wifi:[0m
[0;36m[D][esp-idf:000][1;31m[wifi][0;36m: Association refused temporarily, comeback time 0 mSec[0m
[0;36m[D][esp-idf:000][1;31m[wifi][0;36m: 
[0m
[0;36m[D][esp-idf:000][1;31m[wifi][0;36m: E (14860) wifi:[0m
[0;36m[D][esp-idf:000][1;31m[wifi][0;36m: Association refused temporarily, comeback time 0 mSec[0m
[0;36m[D][esp-idf:000][1;31m[wifi][0;36m: 
[0m
[0;36m[D][esp-idf:000][1;31m[wifi][0;36m: E (14877) wifi:[0m
[0;36m[D][esp-idf:000][1;31m[wifi][0;36m: Association refused temporarily, comeback time 0 mSec[0m
[0;36m[D][esp-idf:000][1;31m[wifi][0;36m: 
[0m
[0;36m[D][esp-idf:000][1;31m[wifi][0;36m: E (14905) wifi:[0m
[0;36m[D][esp-idf:000][1;31m[wifi][0;36m: Association refused temporarily, comeback time 0 mSec[0m
[0;36m[D][esp-idf:000][1;31m[wifi][0;36m: 
[0m
[0;36m[D][esp-idf:000][1;31m[wifi][0;36m: E (14923) wifi:[0m
[0;36m[D][esp-idf:000][1;31m[wifi][0;36m: Association refused temporarily, comeback time 0 mSec[0m
[0;36m[D][esp-idf:000][1;31m[wifi][0;36m: 
[0m
[0;36m[D][esp-idf:000][1;31m[wifi][0;36m: E (14941) wifi:[0m
[0;36m[D][esp-idf:000][1;31m[wifi][0;36m: Association refused temporarily, comeback time 0 mSec[0m
[0;36m[D][esp-idf:000][1;31m[wifi][0;36m: 
[0m
[0;36m[D][esp-idf:000][1;31m[wifi][0;36m: E (14968) wifi:[0m
[0;36m[D][esp-idf:000][1;31m[wifi][0;36m: Association refused temporarily, comeback time 0 mSec[0m
[0;36m[D][esp-idf:000][1;31m[wifi][0;36m: 
[0m
[0;33m[W][wifi_esp32:530][1;31m[arduino_events][0;33m: Event: Disconnected ssid='GezaNET_IoT' bssid=[5mBA:FB:E4:74:4A:F1[6m reason='Association Failed'[0m
[0;33m[W][wifi_esp32:530][1;31m[arduino_events][0;33m: Event: Disconnected ssid='GezaNET_IoT' bssid=[5mBA:FB:E4:74:4A:F1[6m reason='Authentication Failed'[0m
[0;36m[D][wifi:696]: Retrying with hidden networks...[0m
[0;32m[I][wifi:312]: WiFi Connecting to 'GezaNET_IoT'...[0m
[0;33m[W][wifi:668]: Connecting to WiFi network failed. Are the credentials wrong?[0m
[0;33m[W][wifi_esp32:530][1;31m[arduino_events][0;33m: Event: Disconnected ssid='GezaNET_IoT' bssid=[5mBA:FB:E4:74:4A:F1[6m reason='Auth Expired'[0m
[0;32m[I][wifi:312]: WiFi Connecting to 'GezaNET_IoT'...[0m
[0;33m[W][wifi:652]: Error while connecting to network.[0m
[0;36m[D][wifi:696]: Retrying with hidden networks...[0m
[0;32m[I][wifi:312]: WiFi Connecting to 'GezaNET_IoT'...[0m
[0;33m[W][wifi:668]: Connecting to WiFi network failed. Are the credentials wrong?[0m
[0;32m[I][wifi:312]: WiFi Connecting to 'GezaNET_IoT'...[0m
[0;33m[W][wifi_esp32:530][1;31m[arduino_events][0;33m: Event: Disconnected ssid='GezaNET_IoT' bssid=[5mBA:FB:E4:74:4A:F1[6m reason='Association Leave'[0m
[0;33m[W][wifi:673]: WiFi Unknown connection status 0[0m
[0;36m[D][wifi:696]: Retrying with hidden networks...[0m
[0;32m[I][wifi:312]: WiFi Connecting to 'GezaNET_IoT'...[0m
[0;32m[I][wifi:616]: WiFi Connected![0m
[0;35m[C][wifi:427]:   Local MAC: 08:3A:F2:AC:2F:54[0m
[0;35m[C][wifi:432]:   SSID: [5m'GezaNET_IoT'[6m[0m
[0;35m[C][wifi:435]:   IP Address: 192.168.107.110[0m
[0;35m[C][wifi:439]:   BSSID: [5mBA:FB:E4:74:4A:F1[6m[0m
[0;35m[C][wifi:440]:   Hostname: 'home-assistant-glow-ac2f54'[0m
[0;35m[C][wifi:442]:   Signal strength: -13 dB [0;32m▂▄▆█[0m[0m
[0;35m[C][wifi:446]:   Channel: 11[0m
[0;35m[C][wifi:447]:   Subnet: 255.255.255.0[0m
[0;35m[C][wifi:448]:   Gateway: 192.168.107.1[0m
[0;35m[C][wifi:449]:   DNS1: 192.168.1.120[0m
[0;35m[C][wifi:450]:   DNS2: 9.9.9.9[0m
[0;36m[D][wifi:625]: Disabling AP...[0m
[0;35m[C][web_server:128]: Setting up web server...[0m
[0;35m[C][api:025]: Setting up Home Assistant API server...[0m
[0;32m[I][app:062]: setup() finished successfully![0m
[... 278 lines cut ...]
[0;35m[C][wifi:442]:   Signal strength: -12 dB [0;32m▂▄▆█[0m[0m
[0;35m[C][wifi:446]:   Channel: 11[0m
[0;35m[C][wifi:447]:   Subnet: 255.255.255.0[0m
[0;35m[C][wifi:448]:   Gateway: 192.168.107.1[0m
[0;35m[C][wifi:449]:   DNS1: 192.168.1.120[0m
[0;35m[C][wifi:450]:   DNS2: 9.9.9.9[0m
[0;36m[D][wifi:625]: Disabling AP...[0m
[0;35m[C][web_server:128]: Setting up web server...[0m
[0;35m[C][api:025]: Setting up Home Assistant API server...[0m
[0;32m[I][app:062]: setup() finished successfully![0m
E (36750) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (36750) task_wdt:  - loopTask (CPU 1)
E (36750) task_wdt: Tasks currently running:
E (36750) task_wdt: CPU 0: wifi
E (36750) task_wdt: CPU 1: IDLE
E (36750) task_wdt: Aborting.

abort() was called at PC 0x401047e4 on core 0

Backtrace:0x400839ad:0x3ffbe9dc |<-CORRUPTED

ELF file SHA256: 0000000000000000

Rebooting...
```

klaasnicolaas, 2024-08-03:
Which ESP32 boards do you use?

./Klaas

gezuka77, 2024-08-03:
> Which ESP32 boards do you use?
> 
> ./Klaas

ESP32-WROOM-32D
ESP32-WROOM-32U
ESP32-WROOM-32

btw 

ESP8266MOD boards are also behave the same as D1 mini, so throwing

`[http_request.arduino:132] Stream pointer vanished!
`
after first reboot.

klaasnicolaas, 2024-08-05:
I will test it myself again soon, I don't have an explanation or solution for it now. You could always look if something similar occurs in the ESPHome issues.

./Klaas

Martin97y, 2024-08-07:
> I will test it myself again soon, I don't have an explanation or solution for it now. You could always look if something similar occurs in the ESPHome issues.
> 
> ./Klaas

Hey so any solutions? Im using esp8266mod and I have same issue. [http_request.arduino:132] Stream pointer vanished!

klaasnicolaas, 2024-08-07:
> Hey so any solutions?

Please look upstream for this, it's waiting for a solution at ESPHome.

./Klaas

tramix32, 2024-09-04:
Same problem here.. Im using [this](https://aliexpress.com/item/1005006336964908.html) board with ESP-WROOM-32D chip.
Heres logs
[Esp-glow.zip](https://github.com/user-attachments/files/16861769/Esp-glow.zip)

klaasnicolaas, 2024-09-09:
With the latest release build my ESP32 worked without any issues. Are you still experiencing this issue?

./Klaas

tramix32, 2024-09-10:
@klaasnicolaas for now im using custom build of ha-glow with latest adruino framework. And its working solid for a week.
```yaml
 framework:
    type: arduino
    version: latest
```

closed 2024-10-17 by github-actions[bot]