klaasnicolaas/home-assistant-glow #586: Git not found validation error

tamimology, 2024-09-06:
When I validate the code, I get the below error:

```
INFO ESPHome 2024.8.3
INFO Reading configuration /config/powermeter-glow.yaml...
INFO Cloning https://github.com/klaasnicolaas/home-assistant-glow/@4.2.0
Failed config

packages: [source /config/powermeter-glow.yaml:40]
  
  git is not installed but required for external_components.
  Please see https://git-scm.com/book/en/v2/Getting-Started-Installing-Git for installing git.
  remote_package: 
    url: https://github.com/klaasnicolaas/home-assistant-glow/
    ref: 4.2.0
    files: 
      - components/basis.yaml
      - components/updates.yaml
      - components/status_led.yaml
      - components/pulse_meter.yaml
```

klaasnicolaas, 2024-09-06:
I need a little more information to know what the problem is, looks like you manually created something in ESPHome Dashboard.

./Klaas

tamimology, 2024-09-06:
This is my esphome yaml file

```
---
# Home Assistant Glow
#
# Read your electricity meter by means of the pulse LED on your
# meter, useful if you do not have a serial port (P1).
# © Klaas Schoute
#
substitutions:
  device_name: home-assistant-glow
  friendly_name: Home Assistant Glow
  project_version: "4.2.0"
  device_description: "Measure your energy consumption with the pulse LED on your smart meter - ESP8266 Generic"

  # Webserver credentials
  web_username: 'glow'
  web_password: 'esphome'
  
  # Define the GPIO pins
  pulse_pin: GPIO13
  status_led: GPIO5
  led_pin_red: GPIO2
  led_pin_green: GPIO4

esphome:
  name: '${device_name}'
  comment: '${device_description}'
  name_add_mac_suffix: true
  min_version: 2024.6.0
  project:
    name: "klaasnicolaas.home-assistant-glow"
    version: '${project_version}'

dashboard_import:
  package_import_url: github://klaasnicolaas/home-assistant-glow/home-assistant-glow/esp8266.yaml@main

esp8266:
  board: esp01_1m

packages:
  remote_package:
    url: https://github.com/klaasnicolaas/home-assistant-glow/
    ref: "4.2.0"
    files:
      - components/basis.yaml
      - components/updates.yaml
      - components/status_led.yaml
      - components/pulse_meter.yaml

# Enable logging
logger:

# Allow provisioning Wi-Fi via serial
improv_serial:

# WiFi credentials
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 192.168.1.29
    gateway: 192.168.1.1
    subnet: 255.255.255.0
  # Set up a wifi access point 
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: '${friendly_name}'

# In combination with the `ap` this allows the user
# to provision wifi credentials to the device via WiFi AP.
captive_portal:

# Local Web Server running on port 80
web_server:
  id: esphome_web_server
  version: 3
  port: 80
  auth:
    username: '${web_username}'
    password: '${web_password}'
```

tamimology, 2024-09-06:
I am updating it from v 2.2.0

klaasnicolaas, 2024-09-07:
Try to use v4.2.1 

./Klaas

tamimology, 2024-09-09:
> I need a little more information to know what the problem is, looks like you manually created something in ESPHome Dashboard.
> 
> ./Klaas
> 
> 

Yes
I had the old gliw version installed
If I update the ysml code to the latest I get those errors
If I try to use the guide and install using the blue connect button, it fails to access the usb device. I tried it on Windows, Linux, 2 different laptops and different ports each. Same device access error.

closed 2024-10-17 by github-actions[bot]