klaasnicolaas/home-assistant-glow #588: Failed to load packages. components/updates.yaml does not exist in repository

ChrisLee7, 2024-09-08:
Apologies if i've missed something obvious but i've slept on it and still can't figure it out.
I updated glow to 4.21 using the home assistant popup.

I need to relink it back to esphome addon (as it was previously adopted by the addon), when I click install it fails with the following message:
`INFO ESPHome 2024.8.3
INFO Reading configuration /config/esphome/home-assistant-glow-667c9c.yaml...
Failed config

packages: [source /config/esphome/home-assistant-glow-667c9c.yaml:4]
  
  Failed to load packages. components/updates.yaml does not exist in repository.
  klaasnicolaas.home-assistant-glow: github://klaasnicolaas/home-assistant-glow/home-assistant-glow/esp32.yaml@main`

The YAML for the glow in question is:
`substitutions:
  name: home-assistant-glow-667c9c
packages:
  klaasnicolaas.home-assistant-glow: github://klaasnicolaas/home-assistant-glow/home-assistant-glow/esp32.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
api:
  encryption:
    key: [removed for sharing on github]

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
`
From what I can tell, it's accessing the esp32.yaml fine but not able to access the update yaml... to me, all are in the correct places so I don't understand what esphome's issue is.
The glow itself is working but as home assistant has it through the addon, the info isn't making it into HA. 
I've tried removing the device from esphome addon and readopting but get the same message

klaasnicolaas, 2024-09-08:
Tried to reproduce it, but it works fine for me.

Have you tried to install new firmware on your ESP32 via the website and then adopt it in ESPHome Dashboard?

./Klaas

Ps. Thnx for the donation! ❤️

ChrisLee7, 2024-09-08:
Wow, super fast response Klaas - thanks.

Good to know it's just my setup. 
I tried your suggestion but the result was the same.
Then I uninstalled the esphome addon from my home assistant yellow and reinstalled - seems to have done the trick!

If this happens to anyone else: it's not as daunting as it seems, backup the yaml files for each device but clicking uninstall then reinstalling didn't result in any data loss (other than esphome having to download dependencies etc).

Keep up the great work, this project is awesome!

closed 2024-09-08 by ChrisLee7

Eddie46848, 2024-10-22:
> Wow, super fast response Klaas - thanks.
>
> Good to know it's just my setup. I tried your suggestion but the result was the same. Then I uninstalled the esphome addon from my home assistant yellow and [...]

Thanks, just had the same problem, uninstall and reinstall, updated fine, thanks