klaasnicolaas/home-assistant-glow #604: SmartMeter IR LED?

deanfourie1, 2024-10-06:
I have a smart meter with what appears to be an IR LED.

What are my options here, I am trying to measure it but obviously with a IR LED, im a bit stuck as to how I would do this,.

Picture of the meter below.

Thanks

[image]

klaasnicolaas, 2024-10-06:
There are other projects (don't know the name right now) with which you may be able to read the IR LED, the Home Assistant Glow is NOT suitable for this in any case.

./Klaas

TurtleSwift, 2024-10-09:
There is no pulse / metrology LED on this meter it seems. This project will not be able to help you.

What you see here (on the left - round looking with two holes) is an optical port for communication with the meter, indeed in IR. These are used for configuration by the electricity company and of course reading values. In some cases, you can attach an appropriate device to it and read data without any authentication - see this project https://aquaticus.info/meter.html.

closed 2024-10-16 by klaasnicolaas

gapple, 2024-12-02:
@deanfourie1 The meter has `1000 imp/kWh` on it, so I assume there is a pulse LED too.  Have a look for LEDs other than the two for the optical com port on the front - my meter has an IR LED on the top, which I could verify was pulsing at the expected rate with my phone camera.

I've just ordered parts, so I don't know if a regular LM393 module can detect IR wavelengths too, but there is also "IR Flame 
Detection" modules that look the same except with an IR photodiode, or it might be possible to swap the diode on a regular module.  
From what I understand, photodiodes can detect light of a design wavelength or shorter, and IR photodiodes have a dark coating to reduce visible light but is IR-transparent. If a visible light diode can see longer IR wavelengths, it might need to be set more sensitive and be susceptible to reading false pulses from ambient light.

rbarberop, 2025-01-27:
@deanfourie1 interested if you moved forward somehow with your meter. I have a similar (different model) Elster and looking forward to finding a solution to read the meter in real time.

Besides the IR port, the display in my meter also shows the 'square and arrow' below the kwh... Both of them pulse, so I wonder if there is any relationship between consumption and those pulses.

deanfourie1, 2025-01-27:
> [@deanfourie1](https://github.com/deanfourie1) interested if you moved forward somehow with your meter. I have a similar (different model) Elster and looking forward to finding a solution to read [...]

@rbarberop not yet, kind of put this one on the list for looking at later; My meter is not quite easily accessible.

However, in the manual I did see it stated that there "might" be a twin terminal black that gave a pulse count, so I suggest checking all terminals on your meter to see if that the case, as that would make it super simple.

rbarberop, 2025-01-27:
I've opened a [discussion here](https://github.com/klaasnicolaas/home-assistant-glow/discussions/711) , since the 'blocks & arrow' that I saw in the LCD in your pic prompted me to do a bit more research and looks like they are equivalent to the LED pulses.

spdustin, 2025-03-14:
I also have a smart meter (a different brand, but with the same standard IR data port).

Based on the installer manual I retrieved from the FCC, the IR LED on the right of my meter acts as a simple counter unless a command is sent to it via the IR photodiode on the left. From what I've been able to learn, *most* smart meters with the dual-IR data port are configured by default to act the same way.

In my case, the label states "Kh 10.0". Kh is also known as the "meter constant", and it's the number of watts recorded by the meter for every revolution (for disc-style meters) or every LED flash (for modern meters). For me, that translates to an imp/kWh of 100.

(for others' reference: if your meter states "Kh *n*", that's the meter constant. To determine the imp/kWh, divide 1000 by the meter constant value. In my case, that's `1000 / Kh 10 = 100 imp/kWh`)