klaasnicolaas/home-assistant-glow #615: SSH into glow?

acollien, 2024-10-11:
Had a google, but can't find it - don't know what I don't know with the board (ESP32) and the glow OS.

How do I ssh into the glow over wifi?

I assumed it might be similar to the various raspberryPis re: remote access, but I can't find any documentation to walk a newbie, like me, though getting it setup.

I gather there's no server running by default? or maybe the port is blocked - my attempts to ssh in have come back with connection refused.

Much appreciated

klaasnicolaas, 2024-10-11:
You don't find it because this is not possible at all 😅  What were you planning?

./Klaas

acollien, 2024-10-12:
> You don't find it because this is not possible at all 😅 What were you planning?
> 
> ./Klaas

That would be why then! 😅

I was planning to edit the code directly so I don't have to manually upload config files. Unless there is a way through the UI? 

I am just getting some weird readings from Glow (like today - it's about 4PM, I have been out from 9AM, so nothing using power really - added I have solar, so any power passively used would be covered by the solar 10 times over - yet I am see usage of 10,000KWh... and I know the pulse rate is right (1000) because I used a different pulse meter and it gave accurate readings.) so figured I'd try and play around with the config files via ssh.

klaasnicolaas, 2024-10-12:
Where things go wrong right away is that you use both the daily and total energy entities, while you only use 1 of these 2 in the energy dashboard. You also need to remove the PV generation entities on return to grid and only use 1 of those 2 entities in Solar production.

./Klaas

acollien, 2024-10-13:
> Where things go wrong right away is that you use both the daily and total energy entities, while you only use 1 of these 2 in the energy dashboard. You also need to remove the PV generation entities on return to grid and only use 1 of those 2 entities in Solar production.
> 
> Also keep in mind that the pulse LED may behave differently in combination with solar panels and this will affect the glow measurements.
> 
> ./Klaas

Apologies - I only added daily and total for the screenshots, so you could see the kind of values I'm getting on the sensors.

The solar sensors are coming straight from my GoodWe inverter, so are independent of the meter/Glow and they appear to be giving accurate values for generation.

But the Daily Energy reading, coming from glow, is still very far off - I might use 10kwh in a day, so the numbers I am seeing (in the thousands) can't be right.

And re: removing solar from the return-to-grid: Don't I need the daily solar sensor here, so that it can calculate my net solar usage and net energy used?

klaasnicolaas, 2024-10-13:
> Apologies - I only added daily and total for the screenshots, so you could see the kind of values I'm getting on the sensors.

Don't do it because you are ruining your energy dashboard and it doesn't make it any clearer for me or others what exactly the problem is. Except that I now see that your Daily sensor has a very high value.

> But the Daily Energy reading, coming from glow, is still very far off

You may need to set an [internal_filter](https://glow-energy.io/docs/advanced/firmware_customization#make-customizations) or adjust the [calculation method](https://glow-energy.io/docs/faq/abrupt_consumption_with_solar).

>  removing solar from the return-to-grid

Those entities do not belong there because you do not feed your generated energy directly back into the grid, your house first consumes the necessary energy and the remainder goes into the grid.

acollien, 2024-10-13:
> > Apologies - I only added daily and total for the screenshots, so you could see the kind of values I'm getting on the sensors.
> 
> Don't do it because ........... much energy your house uses that runs through the meter.
> 
> ./Klaas

OK, thank you for the advice. I'll take a look at the filter and calculation method. 

I note that I don't have any 'Edit' button in the ESPHome dashboard, as this [page](https://glow-energy.io/docs/advanced/firmware_customization/#make-customizations) indicates - is this potentially because I am running HA in a docker environment on unraid (and not the OS version of HA)? - so I gather I need to edit the YAML files directly on the SD card? re: not being able to SSH

klaasnicolaas, 2024-10-14:
Can you make a screenshot where the "Edit" button be missing? I can hardly imagine that you don't have it when using ESPHome dashboard.

./Klaas

acollien, 2024-10-14:
No problem - quite possible I am looking in the wrong place, but I've not been able to find any editing options that allow me to edit code/config files. 

On the ESPHome page, clicking the 'Configuration' link only gives me one check-box (to toggle "Allow the device to perform Home Assistant actions.")

On the ESPHome device page, the edit pencil in the top right-hand corner, just lets me edit the area, name, and label etc. for the device.

And on the ESPHome entity page, I can't see anything that can be edited. 

[image]
[image]

klaasnicolaas, 2024-10-14:
This is not the ESPHome dashboard but an overview page of the ESPHome integration.

If you are running Home Assistant OS then you should use the add-on, but since you are running everything in separate Docker containers you will need to run an ESPHome container alongside it.

./Klaas

acollien, 2024-10-15:
> I would advise you to delve a little further into how ESPHome works 😉

Sound good - can you point me in the direction of where I can learn more re: docker version etc? as I couldn't find the info I needed here: https://glow-energy.io/

The other other dash I couldn't find is the frontend for glow, but that doesn't have any edit buttons either.

(noticed the time sync is a hour forward too - but that's not so important)

[image]

klaasnicolaas, 2024-10-15:
There is plenty of information available on the [ESPHome](https://esphome.io/) website and [community forum](https://community.home-assistant.io/). This information is not on the Glow website, because I assume that someone already has a working ESPHome and if not, consults the official docs.

- https://community.home-assistant.io/t/an-introduction-to-esphome/709693
- https://esphome.io/guides/getting_started_command_line.html

> The other other dash I couldn't find is the frontend for glow

You must have found it, considering you took a screenshot of it 😉 

./Klaas

closed 2024-10-15 by acollien