klaasnicolaas/home-assistant-glow #617: Upsy Desky documentation license/attribution

tjhorner, 2024-10-19:
Hello! First I want to say that this is a really cool project; I love the ingenuity behind the solution you came up with :)

I noticed that some content in the documentation is derived from my own project, [Upsy Desky](https://upsy-desky.tjhorner.dev/).

Here are some examples from various pages:

[image]

[image]

[image]

And the [`EspInstallButton` component](https://github.com/klaasnicolaas/home-assistant-glow/blob/develop/docs/src/components/EspInstallButton.tsx) appears to be [derived as well](https://github.com/tjhorner/upsy-desky-docs/blob/main/src/components/EspInstallButton.tsx).

This is of course all well and good, and I'm super glad you found these parts useful — after all, that's why I made them open source in the first place! But the documentation content is [licensed under CC BY-NC-SA 4.0](https://github.com/tjhorner/upsy-desky-docs/blob/main/LICENSE), so attribution and redistribution under the same license is required for this derived content.

I would really appreciate a notice being placed at the bottom of pages that contain content derived from the Upsy Desky docs. The following would suffice:

> This page contains content derived from the [Upsy Desky documentation](https://upsy-desky.tjhorner.dev/) by [TJ Horner](https://github.com/tjhorner). The derived content is licensed under [CC BY-NC-SA 4.0](https://creativecommons.org/licenses/by-nc-sa/4.0/deed.en).

As far as I can tell, the following pages contain derived content:

- https://glow-energy.io/docs/getting-started
- https://glow-energy.io/docs/configuration
- https://glow-energy.io/docs/advanced/firmware_customization

Thank you!

klaasnicolaas, 2024-10-19:
Your repository was a very nice starting point to set up a documentation website for esphome device. Especially how you added the firmware install button 😄 

I'm fine with adding your notice, but at the same time it is my preference (and was also on the planning) to rewrite the texts. Except for the page where the "EspInstallButton" is also used. 

./Klaas

tjhorner, 2024-10-22:
Thanks! You can continue using the `EspInstallButton` component as-is, we can just consider it MIT since it's technically code and not documentation 🙂

tjhorner, 2024-10-22:
@klaasnicolaas I was reviewing the changes you made in commit e196a3ce38ff46b65e9b6428f34d6f901a9bd362, and it was my understanding that you were going to rewrite some of the derived content from scratch, but it seems like the existing text was just paraphrased with the structure remaining largely the same.

This is still a derivative work and requires attribution under CC BY-NC-SA. Could you please add the attribution notice to these pages as well?

klaasnicolaas, 2024-10-22:
I had hoped that this would be enough....

Somewhere you have to draw the line to where it still looks like your work or that it deviates too much and is no longer under your license. In the end we are all describing the same steps that are the same for every ESPHome device project in some way.

I will restructure again, but this time more from scratch.

./Klaas

klaasnicolaas, 2024-10-24:
Closing this issue.

**TL;DR**: The documentation for the Glow has been significantly rewritten and should no longer qualifies as derivative work. Any remaining similarities pertain to functional, non-copyrightable information, so the CC BY-NC-SA 4.0 license is no longer applicable.

---

The following changes have been made:
- **Text rewritten**: the entire docs has been rewritten, including new phrasing, structure, tone and explanations, making it substantially different from the original content.
- **New content added**: additional explanations, examples, and structures have been introduced, which are unique to this project and contribute to its originality.

For any remaining similarities between the 2 projects, these relate to **functional and technical information**. Under copyright law, **facts, processes, methods, and systems**, such as the technical steps required to set up a device or configure the ESPHome device are not protected. In this case, the documentation describes **factual technical processes** such as setting up an device or rename it in Home Assistant. These universal steps cannot be copyrighted and fall within the public domain.

The **CC BY-NC-SA 4.0 license** only applies to the **specific expression**, not functional instructions. Since the remaining similarities are purely technical and no longer literally match, the license is not applicable anymore to these parts of the documentation. 

./Klaas

closed 2024-10-24 by klaasnicolaas

tjhorner, 2024-10-25:
Sure, let's consider this resolved. I'd like to leave you with my two cents, though.

Based on how you approached this situation, I don't think you really understand the spirit of the license and open source ethos. You seem to be focused on complying with the letter of the license rather than understanding _why_ I applied it in the first place.

To help you understand my perspective, let's take a look at just some of the decisions I had to make while writing this documentation:

- Which information would be most helpful to first-time users?
- What order should the information be presented in?
- Are there any important common snags that should be called out?
- What format would make the most sense (a list, paragraph, or something else?)

All of the answers were based on my own experience, research, and user feedback. I spent a lot of my own time and effort into making sure the documentation is easy to follow for beginners and advanced users alike. It results in my own "unique expression" of the information as presented in my original work. The value of my work is not in the exact arrangement of the words I used, but also the labor I spent making all the small decisions that make it flow well.

---

The open source community operates under a social contract that is approximately the following:

> We enjoy making cool stuff and sharing it with each other. The work we do should benefit the entire community and enable others to create similarly cool things. In return, we expect others to recognize our contributions and share modifications in the same spirit.

This is a very important set of principles that foster the success of many great projects that we rely on every day like Home Assistant, OctoPrint, Kubernetes, and many more. If this social contract is broken, it invites distrust and toxicity into the entire community; open source only works if everyone plays their part. I played mine by releasing my content openly. But if that's not reciprocated, I am significantly less likely to release my work openly in the future; I assume that is a sentiment shared by many.

We enforce this social contract through a set of open licenses with cryptic legalese. Typically we don't ever have to look at these licenses in much detail because we are operating on the assumption of good faith. We don't want to enforce these licenses, they are simply there as a legal CYA. But if we start to dissect the minutiae of this legalese and lose sight of the intention behind it, we are no better than patent trolls or shady companies exploiting loopholes for profit.

By your own admission, my work was a "very nice starting point" for your own. It's very insulting to me that you are not willing to even add a small acknowledgement of this fact, and would rather spend effort to twist my words until you believe they are _technically_ no longer considered a derivative under the license I chose.

As an open source contributor, I enjoy sharing my work with the world in hopes that others can build upon it and make something amazing, as you did here. All I ask in return is recognition so that my effort is not in vain. Now you may understand why you taking my words, paraphrasing them, claiming them as your own, then telling me _I should take it as a compliment_ feels like a massive slap in the face.

---

You originally copied my documentation effectively word-for-word [back in January of this year](https://github.com/klaasnicolaas/home-assistant-glow/pull/333/files). At this point in time, my repo didn't have a license (the CC license was added in April), which means the only legal way for you to do this would be to contact me — which you did not do. This went unnoticed for 9 months until I happened to catch it by happening to look at HA Glow's docs and recognizing them as my own, then investigating.

Luckily I am not the litigious type, but this type of flagrant disregard for copyright law can result in pretty big consequences if the source author is a large corporation or is otherwise well-funded. If I wanted to be super litigious (I don't, to be clear), I could hire a lawyer to sue for damages because you were using my content without a proper license and financially benefitted from it via sponsorships and Ko-Fi.

---

I feel that I was incredibly reasonable and polite during our interaction here, yet it's quite appalling that I never even received something as small as an apology for your illegal usage of my content for nine months.

I have great respect for those that create open source projects for the entire community to enjoy, including yourself. I know how difficult and time-consuming it is to maintain a project like this since I do so myself. I am sharing this feedback with the hope that you take it to heart and improve how you conduct yourself with situations like these, and to avoid legal/financial/social risks in the future.

Have a good rest of your week.

jnijhof, 2024-10-25:
@tjhorner this will not hold in court since you agreed on https://docs.github.com/en/site-policy/github-terms/github-terms-of-service#5-license-grant-to-other-users

So this is a good learning moment for everyone as well to always start with a license you agree with for your public repositories.

jnijhof, 2024-10-25:
@tjhorner I totally agree on the ethos part, it's just a reminder for everyone reading this to add a license to your open source projects from the start.

klaasnicolaas, 2024-10-25:
In hindsight, I should have paid more attention in January and maybe handled it differently and now I understand better how you look at it. In the end, we are all programmers who want to make cool stuff and not lawyers who know everything about how licenses should be applied and the limits of copyright.

An attribution would have been nicer from me at the time, my apologies that this did not happen. I will certainly take your feedback to heart and let it be a wise lesson for everyone.

./Klaas

PR #623 (2024-10-21): Add attribution for derived content from Upsy Desky documentation
SSIA

Closes: #617 

## Summary by CodeRabbit

- **Documentation**
	- Updated the "Getting Started" guide with additional content and a note regarding the source of some material.
	- Added attribution for content derived from Upsy Desky documentation, including licensing information.