klaasnicolaas/home-assistant-glow #673: Boot loop if internet access is blocked

1337-gitter, 2024-12-01:
On startup, the glow tries to access glow-energy.io (I assume to check for firmware updates). On my network, IOT devices like the glow are located within a separate VLAN which doesn't have access to the internet. In this situation, the glow performs constant resets and cannot be used at all.
I can resolve the situation if I grant internet access to the glow. In this case, everything works like a charm.

Is there a way to disable the connection to glow-energy.io or would it be possible to make it more robust?

Kind regards,
André

klaasnicolaas, 2024-12-01:
I would keep it easy for yourself and give the glow access to the internet, so that you continue to receive firmware updates.

An alternative is to create a custom firmware for your glow. I'm not sure if just adopting is enough to disable the `http_request` update or if you actually need to create your own YAML config from the ground up and customize it.

./Klaas

1337-gitter, 2025-01-08:
I would be glad to receive an answer to my questions.

closed 2025-03-11 by github-actions[bot]