klaasnicolaas/home-assistant-glow #725: Where can I define pins?

HybridZach, 2025-02-15:
Hello
I am wanting to use this project myself but the pins are slightly different with the hardware I currently have. I am using the m5stack atom echo (which uses esp32) and the m5stack unit light (which uses lm393). 

I have seen the custom firmware documentation and looking at each yaml for example  pulse meter yaml  uses variable `pin: ${pulse_pin}`

Please ignote my ignorance I am super new to thsi, is there a way to define these pins for the sensor and RGB light inside the atom in the esphome web component? or is there a config file which defines these variables? or do I simply manually fill them out?

HybridZach, 2025-02-16:
Okay nvm lmao I found it - home-assistant-glow/esp32.yaml

HybridZach, 2025-02-19:
Okay nvm that didn't quite work. I tried to edit the yamls and it wasn't really working. How do I go about customizing it purely to change the pins it reads from correctly? Since it still tries using the remote files? Can I use !extend somehow like it says on your docs?

klaasnicolaas, 2025-03-22:
Extends might not work, because there is no ID defined for pins. Easiest is to copy all yaml code (including components) and compile this as separate YAML config in ESPHome dashboard and install it on your ESP.

./Klaas

closed 2025-04-29 by github-actions[bot]